pointhi/HighPower-Mechaduino #1: Current rating

SloMusti, 2016-10-23:
I believe 6A current rating at 48V is completely sufficient for most applications the board would be used with NEMA 23 motors.

pointhi, 2016-10-23:
I also think so. But the dimension for 10A does requires only minimal changes (I have to define the current sensing a bit wider).

If the finished board can drive 10A permanently has to be tested. I think it's I think mostly defined how wide I can make the copper lines and if the Capacitors are big enough designed. But to have the possibility to drive some peak doesn't sound like a bad idea.

3eggert, 2019-01-15:
In the datasheet is written:
Accurate angle measurements of around ±2° ... 4° after calibration at room temperature
seems not usable for this application to me