pointhi/HighPower-Mechaduino #3: Consider using RPi Zero

SloMusti, 2016-11-02:
Based on the 4EUR cost, RPi zero may be a very good choice: https://www.raspberrypi.org/products/pi-zero/ please consider that.

pointhi, 2016-11-02:
I don't think so.

One problem is we would then likely run an OS, which could conflict with the tight Real time requirements such a controller has. Another problem is that RPI zero is not usable for projects which required bigger quantities of them, because they are normally limited to 1 pice per customer.

SloMusti, 2016-11-02:
Regarding supply-chain the volumes will likely be greater after a while. Please check if of if not real time algorithms can be built on RPi zero, hopefully someone wrote about this.