pointhi/HighPower-Mechaduino #17: Have you tested the board?

mzarnowski, 2017-07-25:
Hi, 
Tried to find any contact info to send you a private message, but couldn't find any thus I'm writing this post.

Have you managed to test current layout of the board? I'm in need of such controller and just wondered if it's working or are there any serious issues that you're aware of?

Let me know if I can help in further development.

Cheers,
Maciej

pointhi, 2017-07-25:
I have some documentation about the first prototype which should give you a good overview of the current status:

https://github.com/pointhi/HighPower-Mechaduino/blob/master/v0.1_BUILD_LOG.md

I found some little layout problems. One is a missing connection which was fixed with some wire on the prototype, the other one were swapped pins in the schematic which didn't correspondent with the arduino pin names.

We started running a modfiied Mechaduino Firmware a few days ago, and it looks good for now. But there seems to be some communication issue with USB, which is not resolved yet (The calibration table given by the firmware has missing values)

radu103, 2018-03-01:
Is it working now ?

pointhi, 2018-03-03:
No fixes for the PCB were pushed to the repo at the moment. The Prototype works so far, but has some weird behaviour between the two output drivers. Hopefully only a partly hardware defect of a MosFet.

The software runs to the extend to have single stepping, constant velocity, constant tor queue successfully working to some extend.

3eggert, 2019-01-15:
Is there still development happening? If not, where there blocking problems or did it only loose importance?

pointhi, 2019-01-16:
we built a second prototype which has the same stepping behavior. Our current assumption is that it is a problem with the current limiters of the two H-Bridges which causes different current-limits to be enforced.

There is still interest in completing the project, but until the issues are solved for full/half-step operation the system is not really ready in my point of view. Unfortunately, I don't have too much time to work on it as well (and other projects as well)

Bastler100, 2019-06-09:
Hallo pointhi,
ich würde mir gerne so einen High-Power-Mechaduino nachbauen. Gibt es eine Version von der Du weißt das sie funktioniert? Hardware und Firmware?  Bei meinen Selbstbauversuchen von Fräsmaschinen habe ich immer in einer Achse Schrittverluste, die will ich in den Griff bekommen.
Vielen Dank für deine Antwort
Hans

pointhi, 2019-06-11:
@Bastler100 no progress so far. Using it only to prevent step losses should work, but the hardware still does not work as seamlessly as I would like. Furthermore, the pcb still misses the bugfixes.

Bastler100, 2019-06-12:
Hallo Thomas,danke für deine Antwort. Ich will dir nicht auf den Wecker gehen, aber eine Frage:Wäre ein Lösungsansatz für einen stärkeren Controller nicht den Motortreiber 
gegen einen eingangssignalkompatiblen Motortreiber oder Gatetreiber 
auszutauschen? Dann könnte man doch die Software unverändert verwenden?Vielen Dank für deine Antwort Hans

Verschickt: Di, 11. Jun. 2019 8:59

pointhi, 2019-06-13:
changing the motor controller is definitly a solution for the problems. Ideally moving from two seperated H-bridges to one integrated motor controller which still allows us to controll currents of both H-bridges separatly.

pointhi, 2019-06-13:
This controller could work: https://www.allegromicro.com/en/Products/Motor-Driver-And-Interface-ICs/Bipolar-Stepper-Motor-Drivers/A3985.aspx

One advantage is that the current is set via internal DAC (no longer disturbances due to analog lines from mcu -> motor controller including ground lines back

Would be nice to add a list of possible controllers

pointhi, 2019-06-13:
Another alternative: https://www.trinamic.com/products/integrated-circuits/details/tmc5160/

Advantage would be that we could offload calculation requirements to the controller because it already handles most of the stuff itself. Disadvantage is that we cannot directly control the current anymore per line, but we get a good control-loop instead.

Bastler100, 2019-06-13:
Hallo Thomas,danke für deine Antwort. Mein Plan war, zuerst die Außenmaße der Platine auf ca 60 x 60mmpassend zu NEMA23 Motore zu vergrössern und anstatt des A4954 Motortreibers zwei A4950 
Motortreiber zu benutzen. Die wären ja Ein- und Ausgangssignalkompatibel und müssten vonder Stromstärke für die 23er ausreichen. Sollte das funktionieren würde ich die Platine für die 
NEMA34 bzw 42 vergrössern und als Motortreiber den DRV8402 nehmen.
MfG
 Hans

pointhi, 2019-06-13:
I created a wiki page to collect possible solutions: https://github.com/pointhi/HighPower-Mechaduino/wiki

annhan, 2020-04-03:
any update for this project

pointhi, 2020-04-04:
@annhan no updates for now.

WesleyBuck, 2020-05-11:
@pointhi Is there anyway we can assist in getting the project moving along?